RobertCNelson/netinstall #39: Kernel 3.18.3 crashes when formatting iSCSI drive

Powersurge, 2015-01-20:
First of all, I must say this is an awesome utility!  I love it!  Debian installed without a hitch.

One issue that I am having is that I cannot figure out how to copy an older kernel image to my SD card.  I even tried specifying the --deb-file tag, but that just causes my Wandboard not to boot.  I believe that the latest kernel 3.18.3 will not work with iSCSI because it crashes when I try to format.  I have a SheevaPlug running Wheezy using kernel 3.0.2 and I can format the same iSCSI drive just fine. Any suggestions you can make would be greatly appreciated.

RobertCNelson, 2015-01-20:
For i.mx6's, give the "--use-beta-kernel" flag, 3.18.x is very unreliable on my wand's, i've already switched them to v3.19-rc4/5. However v3.19-rc5 is broken on omap's, so i haven't pushed v3.19-rc5 over 3.18.x in the stable channel yet.
Regards,

Powersurge, 2015-01-21:
Thanks Robert!  I added the "--use-beta-kernel" flag and everything is working perfectly now.