RobertCNelson/netinstall #40: Tried this on wandboard quad c1

spinza, 2015-02-16:
I does not start up.
I ran this from an ubuntu 14.04 host.  
http://pastebin.com/t36P9VNi

RobertCNelson, 2015-02-16:
@spinza and what does the serial port show?

spinza, 2015-02-16:
I started going through your full build instructions and then I saw the netinstall method in the comments.  I've done the full build before, but prefer to skip full build process.   

The image on the card before I started was working but I wanted to start with a clean image.  My wandboard had been gathering dust and thought I'd try again to get it up and running.  From what I recall the spdif light is a good proxy for whether it's booting.  I recall that the light goes on and then goes off if it's happy (and the existing image did so).  In this case the light stays on.

I don't have serial access?  I've managed without it before.  Do I need it?  I also tried the above with the default debian image.  Same result.

spinza, 2015-02-16:
The only thing I can think of is that I'm trying it with a different hdmi screen than before.  Might that affect things?

spinza, 2015-02-16:
Oh and the other thing is that the there is an error in the pastebin with regard to formatting the partition.  I'm at a different location now so can't access the pastebin or the original dump but you can see it.  It complains about the device being in use.  I wasn't sure if this was expected?

spinza, 2015-02-16:
@RobertCNelson sorry didn't @ you?

RobertCNelson, 2015-02-16:
No it works fine..  After i made sure to tie the gnd off on the serial cable. (terrible board design)  The hdmi came up with the default video parameters and the install began as normal.

Regards,

spinza, 2015-02-16:
It was the monitor.

closed 2015-02-16 by spinza