RobertCNelson/netinstall #43: latest Netinstall method is broken

anish198519851985, 2015-04-14:
I have tried netinstall method to flash my mmc card sometime back and it worked like a charm. However when i tried today i got the below problem and it just exits without doing anything.

command used: sudo ./mk_mmc.sh --mmc /dev/mmcblk0 --dtb omap3-beagle-xm --distro trusty

It bails out after below two lines:
Downloading Device's Kernel Image
-----------------------------
2015-04-13 18:36:36 URL:https://rcn-ee.net/deb/trusty-armhf/LATEST-armv7 [250/250] -> "/tmp/tmp.xQvYrTU4dP/dl/LATEST-armv7" [1]
2015-04-13 18:36:36 URL:https://rcn-ee.net/deb/trusty-armhf/v3.19.3-armv7-x4/ [1036/1036] -> "/tmp/tmp.xQvYrTU4dP/dl/index.html" [1]

It just exits after this. I debugged and found it exits in mk_mmc.sh line number 219. I deleted my tmp folder also but it didn't help. Please let me know if i am doing something wrong.

RobertCNelson, 2015-04-14:
```
git pull
```
I've been migrating from a vps (rcn-ee.net) to dedicated server (rcn-ee.com). Git head of 'NetInstall" has this change.
Regards,

anish198519851985, 2015-04-14:
https://github.com/RobertCNelson/netinstall/commit/470a26f393b054f12be015710961a19c4ad5c244 after this commit soemthign is broken. I tested before this and it is downloading but it is in progress now.

anish198519851985, 2015-04-14:
Oh sorry. I did git pull and it seems to be progressing. Let me close this issue once netinstall method works again.

RobertCNelson, 2015-04-14:
Btw, right now you can also safely use:
```
--use-beta-kernel
```
To get v4.0.0, as i had one more build fix for wheezy, so i didn't push it the "stable" branch thus going over 3.19.x. Since it built fine, when v4.0.1 comes out, it'll automaticly be pushed over 3.19.x ;)
Regards,

anish198519851985, 2015-04-14:
It seems to be not booting. I think i have to wait for 10 to 20 min. I will also use --beta-kernel. Let me update you.

anish198519851985, 2015-04-14:
weird. The logs were not coming from the serial terminal and then i checked my other uEnv.txt and replaced the current uEnv.txt with below command and it worked (atleast i got the logs and currenlty installation is in progress, using serial terminal to install ubuntu):
console=ttyO2,115200n8 root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait fixrtc quiet 

In the current netinstall method uEnv.txt was 
console=tty0 root=/dev/ram0 rw

RobertCNelson, 2015-04-14:
Tested it this morning, everything works like expected.
As you didn't force serial mode "--serial-mode" i can only assume your hdmi monitor might have failed.
Try un-commenting:
https://github.com/RobertCNelson/netinstall/blob/master/hwpack/omap3-beagle-xm.conf#L30
(you'll probally have to do a "git pull" as i just added ^)
Regards,

anish198519851985, 2015-04-14:
I had addeded in uEnv.txt this line as below but it didn't work.
--serial-mode console=tty0 root=/dev/ram0 rw
I don't have hdmi monitor and i want to install it using serial mode. Do you think what i did above in uEnv.txt is the right thing to do?

RobertCNelson, 2015-04-14:
If you want --serial-mode, you have to use this:
```
sudo ./mk_mmc.sh --mmc /dev/mmcblk0 --dtb omap3-beagle-xm --distro trusty --serial-mode
```