RobertCNelson/netinstall #44: Starting kernel ... out of time

fojianfenshuo, 2015-04-14:
the serial tool screen :

U-Boot SPL 2015.04-rc5-00001-g47cb017 (Apr 01 2015 - 13:14:54)
SPL: Please implement spl_start_uboot() for your board
SPL: Direct Linux boot not active!
reading u-boot.img
reading u-boot.img

U-Boot 2015.04-rc5-00001-g47cb017 (Apr 01 2015 - 13:14:54), Build: jenkins-github_Bootloader-Builder-133

OMAP36XX/37XX-GP ES1.2, CPU-OPP2, L3-200MHz, Max CPU Clock 1 Ghz
OMAP3 Beagle board + LPDDR/NAND
I2C:   ready
DRAM:  512 MiB
NAND:  0 MiB
MMC:   OMAP SD/MMC: 0
*** Warning - readenv() failed, using default environment

Beagle xM Rev C
No EEPROM on expansion board
No EEPROM on expansion board
Die ID #2adc00029ff8000001578ca01002d017
Net:   usb_ether
Error: usb_ether address not set.

Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
Checking for: /uEnv.txt ...
Checking for: /boot/uEnv.txt ...
reading /boot/uEnv.txt
185 bytes read in 6 ms (29.3 KiB/s)
Loaded environment from /boot/uEnv.txt
Checking if uname_r is set in /boot/uEnv.txt...
Running uname_boot ...
loading /boot/vmlinuz-current ...
reading /boot/vmlinuz-current
5189432 bytes read in 323 ms (15.3 MiB/s)
loading /boot/dtbs/current/omap3-beagle-xm.dtb ...
reading /boot/dtbs/current/omap3-beagle-xm.dtb
63096 bytes read in 27 ms (2.2 MiB/s)
loading /boot/initrd.img-current ...
reading /boot/initrd.img-current
39627728 bytes read in 2418 ms (15.6 MiB/s)

Installer for [trusty-armhf] is using the Serial Interface

debug: [console=ttyS2,115200n8 root=/dev/ram0 rw] ...
debug: [bootz 0x82000000 0x88080000:25cabd0 0x88000000] ...
Kernel image @ 0x82000000 [ 0x000000 - 0x4f2f38 ]
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Ramdisk to 8da35000, end 8ffffbd0 ... OK
   Using Device Tree in place at 88000000, end 88012677

Starting kernel ...

AND this state lasted a long time(about 20 minutes), and then I upplug the power。
The question is I dont understand the step:
"
Configure the network
usb0: USB net <- (usually the OTG port)
eth0: USB net <- (usually the smsc95xx adapter on the BeagleBoard and PandaBoard)
wlan0: Wifi <- Your USDB-Wi-Fi device.. 
"
so I skipped this step ,did it matter?

RobertCNelson, 2015-04-14:
Sorry, should be fixed with:
https://github.com/RobertCNelson/netinstall/commit/9eac66fd86e4abae93360d263c6cc90bb1d80345

In the kernel, i had to revert from the new "ttySx" generic omap serial driver back to the older "ttyOx" omap driver.

Regards,

RobertCNelson, 2015-04-14:
Tested this morning, looks good on my xm, please re-open if the issue comes up again.

closed 2015-04-14 by RobertCNelson

fojianfenshuo, 2015-04-15:
It has been solved, thanks.