RobertCNelson/netinstall #47: Unable to configure netinstall

mickaelv, 2015-09-24:
After Pulling the last mk_mmc script, It can't configure the install properly. The script can't find /tmp/tmp.baIVEcx032/bootscripts/*.cmd but ls command return :
>  ls  /tmp/tmp.baIVEcx032/bootscripts/*.cmd
> /tmp/tmp.baIVEcx032/bootscripts/netinstall.cmd	/tmp/tmp.baIVEcx032/bootscripts/normal.cmd

# System 
> Linux mwl-dev 4.0.0-1-amd64 #1 SMP Debian 4.0.2-1 (2015-05-11) x86_64 GNU/Linux

# Install log 
> root@mwl-dev:~/netinstall# ./mk_mmc.sh --mmc /dev/sdd --dtb am335x-boneblack --distro jessie 

-----------------------------

Script Version git: 97f2972
-----------------------------

Downloading Device's Bootloader
-----------------------------
2015-09-24 08:51:11 URL:https://rcn-ee.com/repos/bootloader/latest/bootloader-ng [7226/7226] -> "/tmp/tmp.baIVEcx032/dl/bootloader-ng" [1]
2015-09-24 08:51:13 URL:https://rcn-ee.com/repos/bootloader/am335x_evm/MLO-am335x_evm-v2015.10-rc3-r6 [65476/65476] -> "/tmp/tmp.baIVEcx032/dl/MLO-am335x_evm-v2015.10-rc3-r6" [1]
SPL Bootloader: MLO-am335x_evm-v2015.10-rc3-r6
--2015-09-24 08:51:13--  https://rcn-ee.com/repos/bootloader/am335x_evm/u-boot-am335x_evm-v2015.10-rc3-r6.img
Résolution de rcn-ee.com (rcn-ee.com)… 45.33.2.10, 2600:3c00::f03c:91ff:fe37:6ad5
Connexion à rcn-ee.com (rcn-ee.com)|45.33.2.10|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 314856 (307K)
Sauvegarde en : « /tmp/tmp.baIVEcx032/dl/u-boot-am335x_evm-v2015.10-rc3-r6.img »

u-boot-am335x_evm-v2015.10- 100%[=============================================>] 307,48K   257KB/s   ds 1,2s   

2015-09-24 08:51:15 (257 KB/s) — « /tmp/tmp.baIVEcx032/dl/u-boot-am335x_evm-v2015.10-rc3-r6.img » sauvegardé [314856/314856]

UBOOT Bootloader: u-boot-am335x_evm-v2015.10-rc3-r6.img

Downloading Device's Kernel Image
-----------------------------
2015-09-24 08:51:15 URL:https://rcn-ee.com/repos/latest/jessie-armhf/LATEST-omap-psp [111/111] -> "/tmp/tmp.baIVEcx032/dl/LATEST-omap-psp" [1]
-----------------------------
Kernel Options:
ABI:1 LTS 4.1.8-bone16
ABI:1 STABLE 3.8.13-bone77
ABI:1 TESTING 4.2.1-bone2
ABI:1 EXPERIMENTAL 4.3.0-rc2-bone0
-----------------------------
LTS: --use-lts-kernel
STABLE: --use-stable-kernel
TESTING: --use-testing-kernel
EXPERIMENTAL: --use-experimental-kernel
-----------------------------
--2015-09-24 08:51:15--  https://rcn-ee.com/repos/debian/pool/main/l/linux-upstream/linux-image-3.8.13-bone77_1jessie_armhf.deb
Résolution de rcn-ee.com (rcn-ee.com)… 45.33.2.10, 2600:3c00::f03c:91ff:fe37:6ad5
Connexion à rcn-ee.com (rcn-ee.com)|45.33.2.10|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 13997376 (13M) [application/x-debian-package]
Sauvegarde en : « /root/netinstall/dl/jessie-armhf/linux-image-3.8.13-bone77_1jessie_armhf.deb »

linux-image-3.8.13-bone77_1 100%[=============================================>]  13,35M   507KB/s   ds 34s    

2015-09-24 08:51:50 (403 KB/s) — « /root/netinstall/dl/jessie-armhf/linux-image-3.8.13-bone77_1jessie_armhf.deb » sauvegardé [13997376/13997376]

Using Kernel: linux-image-3.8.13-bone77_1jessie_armhf.deb

Downloading NetInstall Image
-----------------------------
--2015-09-24 08:51:50--  http://ftp.debian.org/debian/dists/jessie/main/installer-armhf/20150422+deb8u2/images/netboot/initrd.gz
Résolution de ftp.debian.org (ftp.debian.org)… 130.89.148.12, 2001:610:1908:b000::148:12
Connexion à ftp.debian.org (ftp.debian.org)|130.89.148.12|:80… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 12162815 (12M) [application/x-gzip]
Sauvegarde en : « /root/netinstall/dl/jessie-armhf/initrd.gz »

initrd.gz                   100%[=============================================>]  11,60M   570KB/s   ds 21s    

2015-09-24 08:52:11 (567 KB/s) — « /root/netinstall/dl/jessie-armhf/initrd.gz » sauvegardé [12162815/12162815]

md5sum of NetInstall: c67ede59f93a0b0f6f70e5ae7ac27659
**sed: impossible de lire /tmp/tmp.baIVEcx032/bootscripts/*.cmd: Aucun fichier ou dossier de ce type**

**Translation : sed : cannot access /tmp/tmp.baIVEcx032/bootscripts/*.cmd:  No such file or directory**

Script ended here. 
Please let me know if i am doing something wrong. Thanks.

RobertCNelson, 2015-09-24:
Sorry about that, i got a little over eager from the shell script analysis, fixed in: https://github.com/RobertCNelson/netinstall/commit/66757c1109666ebc9f723946cd9568c4382b944c

and just tested:  sudo ./mk_mmc.sh --mmc /dev/sdd --dtb am335x-boneblack --distro jessie

mickaelv, 2015-09-24:
Hi Robert,

Thanks for your fast answer and fast correction,

Regards,
Mickaël

closed 2015-09-24 by mickaelv