RobertCNelson/netinstall #48: More informative information in README please

kizzlebot, 2015-10-27:
This repository is referenced from elinux pages but this repo's README and elinux doesn't exactly state what "netinstall" is used for.  I had to try it myself to see that it does what I thought that it did.  Can you please give more general information in the READMEs for these repos?

RobertCNelson, 2015-10-27:
@kizzlebot netinstall = Network Install for a bunch of arm boards

Mausy5043, 2015-11-29:
I have to agree with @kizzlebot: From the README it is not entirely clear to me what the final product actually is/does. Especially given that the BeagleBone Black allows for various ways of running an OS (from eMMC and from uSD). My main question was: Is this a repo that creates a flashable image on the uSD or does the installation have to be run from the uSD?

I started reading `mk_mmc.sh` but the code is soo long that I gave up (sorry). Probably because you want to support so many different hardware platforms.

For reference: I also have several Raspberry Pis for which I use this netinstaller: [https://github.com/debian-pi/raspbian-ua-netinst]. 
This works by me having to push a small image onto an SD and after power up of the Pi it formats the SD and downloads and installs the desired kernel and packages. It even allows for post-install customisation. So I was wondering if this repo is approximately similar.

Anyway, I'm sure to give this repo a try. Thanks and keep up the good work!

RobertCNelson, 2015-11-29:
@Mausy5043 due to the customization of the "debian installer" 

https://wiki.debian.org/DebianInstaller

It is "impossible" for me to write any readme that clearly explains exactly what your final product could be.

All this script does, downloads "debian installer" for a specific distro, wheezy/jessie/stretch trusty/etc, modifies the initrd for our custom kernel, add's the apt repo and makes sure the kernel is properly installed.  It is "completely" up to you to decide everything else, the "only" limitation, the 100Mb "boot" partition must not be modified.

Regards,