RobertCNelson/netinstall #50: repos.rcn-ee.com inaccessible over IPv6

thefinn93, 2015-12-16:
`repos.rcn-ee.com` has an AAAA record, but it does not respond (via http, icmp) over IPv6. The Debian installer doesn't seem to want to fall back to IPv4 upon IPv6 failure. I've configured my local DNS resolver to respond with only the IPv4 address for now, but this is a hack at best and really not ideal

thefinn93, 2015-12-16:
Also, not sure if this is the appropriate place to post about it, if somewhere else is better please direct me there...

RobertCNelson, 2015-12-16:
@thefinn93 

Is there a good way to test ipv6. (i'm stuck on ipv4 here)
With linode, repos.rcn-ee.com should work over ipv6:

```
A/AAAA Records
Hostname	IP Address	TTL	Options
45.33.2.10	Default	Edit | Remove
repos	45.33.2.10	Default	Edit | Remove
www	45.33.2.10	Default	Edit | Remove
2600:3c00::f03c:91ff:fe37:6ad5	Default	Edit | Remove
repos	2600:3c00::f03c:91ff:fe37:6ad5	Default	Edit | Remove
www	2600:3c00::f03c:91ff:fe37:6ad5	Default	Edit | Remove
Add a new A record
```

Unless i have misconfg in apache..

Regards,

thefinn93, 2015-12-16:
It's not an apache issue. I can't ping it over IPv6 either. Perhaps a linode routing issue? My traceroutes seem to be dying at different things named `0.00.0000.ip4.static.sl-reverse.com`:

```
# From home (Comcast)
$ traceroute6 2600:3c00::f03c:91ff:fe37:6ad5
traceroute to 2600:3c00::f03c:91ff:fe37:6ad5 (2600:3c00::f03c:91ff:fe37:6ad5) from 2601:600:8100:4e2:d956:ff08:9241:420f, 30 hops max, 24 byte packets
 1  pfSense.home.finn.io (2601:600:8100:4e2:20a:5eff:fe51:cbb)  412.55 ms  32.798 ms  20.767 ms
 2  2001:558:4082:2b::1 (2001:558:4082:2b::1)  70.405 ms  50.464 ms *
 3  * te-0-1-0-3-sur03.bellevue.wa.seattle.comcast.net (2001:558:a2:bf::1)  35.128 ms  66.437 ms
 4  be-1-sur02.bellevue.wa.seattle.comcast.net (2001:558:a0:f6c0::1)  16.313 ms *  54.428 ms
 5  be-40-ar01.burien.wa.seattle.comcast.net (2001:558:a0:189::2)  63.653 ms  38.301 ms  56.087 ms
 6  he-0-13-0-0-ar01.seattle.wa.seattle.comcast.net (2001:558:a0:18::2)  91.476 ms  79.632 ms  82.95 ms
 7  be-33650-cr02.seattle.wa.ibone.comcast.net (2001:558:0:f769::1)  137.057 ms  71.49 ms *
 8  he-0-11-0-1-pe05.seattle.wa.ibone.comcast.net (2001:558:0:f626::2)  108.584 ms  107.456 ms  113.741 ms
 9  as36351-2-c.seattle.wa.ibone.comcast.net (2001:559:0:c::2)  72.334 ms  44.99 ms *
10  * po3.bbr02.cf01.den01.networklayer.com (2607:f0d0:2:2::11)  130.042 ms  203.379 ms
11  po4.bbr02.eq01.dal01.networklayer.com (2607:f0d0:2:2::a)  112.118 ms  104.427 ms  110.473 ms
12  3700.0000.0000.0000.2000.2000.0d0f.7062.ip6.static.sl-reverse.com (2607:f0d0:2:2::73)  106.006 ms  130.109 ms  206.333 ms
13  5.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:5)  153.828 ms  188.823 ms  139.34 ms
14  0.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:f)  220.865 ms  313.079 ms  307.764 ms
15  * * *
16  * * *
17  * * *
18  * * *
19  * * *
20  * * *
21  * * *
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  * * *
30  * * *

# From Digital Ocean NYC
$  traceroute6 repos.rcn-ee.com
traceroute to repos.rcn-ee.com (2600:3c00::f03c:91ff:fe37:6ad5), 30 hops max, 80 byte packets
 1  2604:a880:800:10:ffff:ffff:ffff:fff2 (2604:a880:800:10:ffff:ffff:ffff:fff2)  0.399 ms 2604:a880:800:10:ffff:ffff:ffff:fff1 (2604:a880:800:10:ffff:ffff:ffff:fff1)  12.142 ms  12.285 ms
 2  2604:a880:800::601 (2604:a880:800::601)  0.337 ms 2604:a880:800::801 (2604:a880:800::801)  0.334 ms  0.293 ms
 3  te2-6.bbr01.tl01.nyc01.networklayer.com (2001:504:17:115::22)  1.052 ms 2604:a880:800::302 (2604:a880:800::302)  0.270 ms  0.287 ms
 4  po1.bbr02.tl01.nyc01.networklayer.com (2607:f0d0:2:2::31)  1.054 ms te2-6.bbr01.tl01.nyc01.networklayer.com (2001:504:17:115::22)  1.092 ms  1.131 ms
 5  po1.bbr02.tl01.nyc01.networklayer.com (2607:f0d0:2:2::31)  1.108 ms po5.bbr02.eq01.chi01.networklayer.com (2607:f0d0:2:2::4)  22.625 ms po1.bbr02.tl01.nyc01.networklayer.com (2607:f0d0:2:2::31)  1.111 ms
 6  po3.bbr01.eq01.dal01.networklayer.com (2607:f0d0:2:2::8)  42.954 ms po5.bbr02.eq01.chi01.networklayer.com (2607:f0d0:2:2::4)  22.001 ms  21.952 ms
 7  * * po3.bbr01.eq01.dal01.networklayer.com (2607:f0d0:2:2::8)  40.342 ms
 8  * * 3.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:3)  43.405 ms
 9  3.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:3)  40.718 ms  40.362 ms 9.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:9)  43.177 ms
10  * * *
11  * * *
12  * * *
13  * * *
14  * * *
15  * * *
16  * * *
17  * * *
18  * * *
19  * * *
20  * * *
21  * * *
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  * * *
30  * * *

RobertCNelson, 2015-12-16:
Closers... it was my ip6tables rule..

validating with: http://ipv6-test.com/validate.php
rcn-ee.com now works
repos.rcn-ee.com still broken..

RobertCNelson, 2015-12-16:
Okay down to apache it looks like:
rcn-ee.com:
```
TraceRoute IPv6 Output:
traceroute to rcn-ee.com (2600:3c00::f03c:91ff:fe37:6ad5), 30 hops max, 40 byte packets
 1  2a02:348:82::1 (2a02:348:82::1)  15.165 ms  0.285 ms  15.144 ms
 2  te0-22.cr1.nkf.as49685.net (2001:4cb8:40b:1::1d01)  1.481 ms  1.508 ms  1.576 ms
 3  bbr01.eq01.ams01.networklayer.com (2001:7f8:1::a503:6351:1)  14.635 ms  14.735 ms  14.710 ms
 4  ec00.0000.0000.0000.2000.2000.0d0f.7062.ip6.static.sl-reverse.com (2607:f0d0:2:2::ce)  80.203 ms  79.671 ms  79.626 ms
 5  po99.bbr01.eq01.wdc02.networklayer.com (2607:f0d0:2:2::42)  79.295 ms  79.600 ms  79.247 ms
 6  po3.bbr02.tl01.atl01.networklayer.com (2607:f0d0:2:2::19)  91.043 ms  91.130 ms  91.357 ms
 7  po7.bbr02.eq01.dal01.networklayer.com (2607:f0d0:2:2::6)  112.184 ms  112.505 ms  113.018 ms
 8  3700.0000.0000.0000.2000.2000.0d0f.7062.ip6.static.sl-reverse.com (2607:f0d0:2:2::73)  111.051 ms  111.725 ms  111.034 ms
 9  3.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:3)  112.315 ms 5.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:5)  110.768 ms 3.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:3)  112.051 ms
10  9.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:9)  110.609 ms  110.702 ms  110.523 ms
11  3.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::2:3)  111.101 ms  111.181 ms  111.340 ms
12  2600:3c00::f03c:91ff:fe37:6ad5 (2600:3c00::f03c:91ff:fe37:6ad5)  110.956 ms  110.920 ms  110.876 ms
```

```
TraceRoute IPv6 Output:
traceroute to repos.rcn-ee.com (2600:3c00::f03c:91ff:fe37:6ad5), 30 hops max, 40 byte packets
 1  2a02:348:82::1 (2a02:348:82::1)  0.334 ms  0.338 ms  0.358 ms
 2  te0-22.cr1.nkf.as49685.net (2001:4cb8:40b:1::1d01)  0.845 ms  0.887 ms  0.915 ms
 3  bbr01.eq01.ams01.networklayer.com (2001:7f8:1::a503:6351:1)  16.410 ms  16.402 ms  16.377 ms
 4  ec00.0000.0000.0000.2000.2000.0d0f.7062.ip6.static.sl-reverse.com (2607:f0d0:2:2::ce)  79.013 ms  79.650 ms  79.143 ms
 5  po99.bbr01.eq01.wdc02.networklayer.com (2607:f0d0:2:2::42)  79.352 ms  78.991 ms  79.093 ms
 6  po3.bbr02.tl01.atl01.networklayer.com (2607:f0d0:2:2::19)  91.307 ms  92.113 ms  91.234 ms
 7  po7.bbr02.eq01.dal01.networklayer.com (2607:f0d0:2:2::6)  116.682 ms  116.637 ms  111.784 ms
 8  3700.0000.0000.0000.2000.2000.0d0f.7062.ip6.static.sl-reverse.com (2607:f0d0:2:2::73)  110.802 ms  110.676 ms  111.704 ms
 9  3.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:3)  111.819 ms 5.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:5)  111.158 ms 3.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:3)  112.171 ms
10  0.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:f)  110.905 ms 9.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:9)  110.613 ms 0.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::1:f)  110.717 ms
11  3.00.0000.ip4.static.sl-reverse.com (2607:f0d0:2:4::2:3)  112.404 ms  112.135 ms  110.929 ms
12  2600:3c00::f03c:91ff:fe37:6ad5 (2600:3c00::f03c:91ff:fe37:6ad5)  110.895 ms  110.861 ms  111.086 ms
```

RobertCNelson, 2015-12-16:
Okay, http://repos.rcn-ee.com/ now seems to work over ip6.. please test. ;)

This was my ip6tables rule...

```
*filter

# Allow all loopback (lo0) traffic and reject traffic
# to localhost that does not originate from lo0.
-A INPUT -i lo -j ACCEPT
-A INPUT ! -i lo -s ::1/128 -j REJECT

# Allow ICMP
-A INPUT -p icmpv6 -m state --state NEW -j ACCEPT

# Allow HTTP and HTTPS connections from anywhere
# (the normal ports for web servers).
-A INPUT -p tcp --dport 80 -m state --state NEW -j ACCEPT
-A INPUT -p tcp --dport 443 -m state --state NEW -j ACCEPT

# Allow inbound traffic from established connections.
-A INPUT -m state --state ESTABLISHED,RELATED -j ACCEPT

# Log what was incoming but denied (optional but useful).
-A INPUT -m limit --limit 5/min -j LOG --log-prefix "ip6tables_INPUT_denied: " --log-level 7

# Reject all other inbound.
-A INPUT -j REJECT

# Log any traffic which was sent to you
# for forwarding (optional but useful).
-A FORWARD -m limit --limit 5/min -j LOG --log-prefix "ip6tables_FORWARD_denied: " --log-level 7

# Reject all traffic forwarding.
-A FORWARD -j REJECT

COMMIT
```

thefinn93, 2015-12-16:
Yep! Seems to work now. Thanks :)

closed 2015-12-16 by thefinn93