RobertCNelson/netinstall #51: Help!!! "Please include /var/log/netinstall.log..."

minhlevn, 2015-12-30:
I want to install Ubuntu into SDCard for beagleboard xm. I used netinstall of RobertCNelson, but when I finished, it announced that "Please include /var/log/netinstall.log from RootFileSystem". And when I inserted my SD card into Beagleboard xm, it booted but at step of Choose a mirror, It messaged "/var/log/......" again. What should I do with this problem? How can I include netinstall.log as the message? Thank you.

RobertCNelson, 2015-12-30:
@minhlevn There is a step by step example here:

https://github.com/RobertCNelson/netinstall/blob/master/test.Ubuntu

Please clarify what stage you got to and what issue your having

Regards,

minhlevn, 2015-12-31:
I finished install with messages:

Finished populating Boot Partition
-----------------------------
mk_mmc.sh script complete
Script Version git: e5934ae
-----------------------------
Reporting Bugs:
https://github.com/RobertCNelson/netinstall/issues
Please include: /var/log/netinstall.log from RootFileSystem

Then I inserted my SDCard into Beagleboard xM. Until "[!!] Choose a mirror of the Ubuntu archive" step, It messaged: "Bad Archive mirror. incorrect mirror specified, mirror is not available; mirror is broken,.... Additional details available in /var/log/syslog or on virtual console 4"

Actually, I didn't fill any name on "Ubuntu archive mirror hostname" or on "Ubuntu archive mirror directory". It hold me on this step.

RobertCNelson, 2015-12-31:
This 
```
"Bad Archive mirror. incorrect mirror specified, mirror is not available; mirror is broken,.... Additional details available in /var/log/syslog or on virtual console 4"
```
could be a bad network connection, or ubuntu.com could be having network problems*..  Please verify which "--distro xyz" option you selected and i'll verify it tomorrow..

Regards,

Note: if it's ubuntu's network problem, there's not much we can really do, except wait till they fix it..