RobertCNelson/netinstall #55: xenial netinstall fails during bootloader installation

mscdex, 2016-11-20:
Here's what I see in syslog after the bootloader installation ("Make the system bootable" step) fails:

```
Nov 20 23:18:32 in-target: flash-kernel is already the newest version (3.0~rc.4ubuntu62).
Nov 20 23:18:32 in-target: 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Nov 20 23:18:34 in-target: Reading package lists...
Nov 20 23:18:34 in-target: 
Nov 20 23:18:34 in-target: Building dependency tree...
Nov 20 23:18:35 in-target: 
Nov 20 23:18:35 in-target: Reading state information...
Nov 20 23:18:35 in-target: 
Nov 20 23:18:37 in-target: u-boot-tools is already the newest version (2016.01+dfsg1-2ubuntu1).
Nov 20 23:18:37 in-target: 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Nov 20 23:18:38 in-target: update-initramfs: Generating /boot/initrd.img-4.4.33-armv7-x13
Nov 20 23:19:20 in-target: flash-kernel: installing version 4.4.33-armv7-x13
Nov 20 23:19:21 in-target: find: ��‘/lib/firmware/4.4.33-armv7-x13/device-tree/��’
Nov 20 23:19:21 in-target: : No such file or directory
Nov 20 23:19:21 flash-kernel-installer: error: flash-kernel failed
Nov 20 23:19:21 main-menu[897]: WARNING **: Configuring 'flash-kernel-installer' failed with error code 1
Nov 20 23:19:21 main-menu[897]: WARNING **: Menu item 'flash-kernel-installer' failed.
```

I'm using the omap4-panda dtb FWIW

RobertCNelson, 2016-11-20:
@mscdex i think this will fix it..

Please do a git pull and retry..

Regards,

closed 2016-11-20 by RobertCNelson

mscdex, 2016-11-21:
That worked. Thanks for the quick response and the quick fix!