RobertCNelson/netinstall #57: Jetson TK1 : partition disk

ch0mik, 2017-01-29:
Hello,

Im now installing the Debian Jessie on my Jetson TK1 via Yours scripts ... its easy way than official Debian Wiki ! (https://wiki.debian.org/InstallingDebianOn/NVIDIA/Jetson-TK1)

I have SATA disk /dev/sda for root filesystems, but I tried with /boot/ for many times ... without boot after reboot.

http://obrazki.elektroda.pl/9951589700_1485648616.png - test with boot on SD (/dev/mmcblk1p1), I test it on first partition on eMMC (/dev/mmcblk0p1). What I should to do under partitioning disks ?

Regards
Pawel

RobertCNelson, 2017-01-29:
@ch0mik the trick is to not alter the initially created boot partition on the microSD.  There's a few scripts on their required by the debian-installer for finalization.  And i have a boot.scr/uEnv.txt script in the boot partition for u-boot to know where to jump to the rootfs:

```
voodoo@b3-jetson-tk1-2gb:~$ uname -r
4.9.5-armv7-lpae-x4
voodoo@b3-jetson-tk1-2gb:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.0 (stretch)
Release:	9.0
Codename:	stretch
```

```
voodoo@b3-jetson-tk1-2gb:~$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda            8:0    0 119.2G  0 disk 
├─sda1         8:1    0     1K  0 part 
├─sda5         8:5    0   3.7G  0 part [SWAP]
└─sda6         8:6    0 115.5G  0 part /
mmcblk0      179:0    0  14.7G  0 disk 
mmcblk0boot0 179:8    0     4M  1 disk 
mmcblk0boot1 179:16   0     4M  1 disk 
mmcblk0rpmb  179:24   0     4M  0 disk 
mmcblk1      179:32   0   3.7G  0 disk 
└─mmcblk1p1  179:33   0   144M  0 part /boot/uboot
```

```
voodoo@b3-jetson-tk1-2gb:~$ ls -lh /boot/uboot/boot/
total 18M
-rw-r--r-- 1 root root 4.7K Jul 31 11:28 boot.scr
drwxr-xr-x 3 root root 1.0K Jan 22 05:32 dtbs
-rw-r--r-- 1 root root  11M Jan 22 05:32 initrd.img-current
-rw-r--r-- 1 root root  354 Jul 31 11:39 uEnv.txt
-rwxr-xr-x 1 root root 6.7M Jan 22 05:32 vmlinuz-current
```

Regards,

ch0mik, 2017-01-29:
Robert,

is it mean : in partitioning disk dialog dont format existing partition   mmcblk1p1 (leave as is was) and use (mount) mmcblk1p1 as /boot/ or as /boot/uboot/  ?

Regards
Pawel

RobertCNelson, 2017-01-29:
@ch0mik yeah, just leave the boot partition the way it's auto-configured.. it'll get added to /etc/fstab later:

https://github.com/RobertCNelson/netinstall/blob/master/lib/debian-finish.sh#L69-L73

Regards,

closed 2017-04-28 by ch0mik