RobertCNelson/netinstall #58: After install : dosnt recognition usb devices (TK1)

ch0mik, 2017-01-30:
Hello,

after install I cant logon to X-Window, Console via keyboard.
lsusb : 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

dmesg : [](https://justpaste.it/12yns)

netinstall.log 

> NetInstall Log:
> fdisk -l...
> cat /boot/uboot/mounts...
> rootfs / rootfs rw 0 0
> none /run tmpfs rw,nosuid,relatime,size=205788k,mode=755 0 0
> none /proc proc rw,relatime 0 0
> none /sys sysfs rw,relatime 0 0
> devtmpfs /dev devtmpfs rw,relatime,size=971008k,nr_inodes=174464,mode=755 0 0
> devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000 0 0
> /dev/sda1 /target ext4 rw,relatime,errors=remount-ro,data=ordered 0 0
> /dev/sda1 /dev/.static/dev ext4 rw,relatime,errors=remount-ro,data=ordered 0 0
> devtmpfs /target/dev devtmpfs rw,relatime,size=971008k,nr_inodes=174464,mode=755 0 0
> /dev/mmcblk1p1 /target/boot/uboot ext2 rw,relatime,block_validity,barrier,user_xattr,acl 0 0
> none /target/sys sysfs rw,relatime 0 0

Ps. - after few minutes I got in dmesg : 

> [  45.996306] vgaarb: this pci device is not a vga device
> [   46.399068] vgaarb: this pci device is not a vga device
> [   46.455819] tegra-hdmi 54280000.hdmi: cannot set audio to 48000 Hz at 85500000 Hz pixel clock
> [  650.270973] vgaarb: this pci device is not a vga device
> 

and Xwindow hangs ...

with Regards
Pawel

RobertCNelson, 2017-01-30:
We looks like we are missing the usb firmware:

```
Jan 30 15:16:42 b3-jetson-tk1-2gb kernel: tegra-xusb 70090000.usb: Direct firmware load for nvidia/tegra124/xusb.bin failed with error -2
Jan 30 15:16:42 b3-jetson-tk1-2gb kernel: tegra-xusb 70090000.usb: failed to request firmware: -2
Jan 30 15:16:42 b3-jetson-tk1-2gb kernel: tegra-xusb 70090000.usb: failed to load firmware: -2
Jan 30 15:16:42 b3-jetson-tk1-2gb kernel: tegra-xusb: probe of 70090000.usb failed with error -2
```

```
root@b3-jetson-tk1-2gb:/home/voodoo# cat /usr/share/initramfs-tools/hooks/board-firmware 
#!/bin/sh -e

if [ "$1" = "prereqs" ]; then exit 0; fi

. /usr/share/initramfs-tools/hook-functions

if [ -d /lib/firmware/ ] ; then
    if [ -f /lib/firmware/nvidia/tegra124/xusb.bin ] ; then
        mkdir -p $DESTDIR/lib/firmware/nvidia/tegra124/
        cp -a /lib/firmware/nvidia/tegra124/xusb.bin $DESTDIR/lib/firmware/nvidia/tegra124/xusb.bin
    fi
fi
```
```
chmod +x /usr/share/initramfs-tools/hooks/board-firmware
```
```
update-initramfs -uk `uname -r`
cp -v /boot/initrd.img-`uname -r`  /boot/uboot/boot/initrd.img-current
reboot
```

```
Jan 30 15:31:06 b3-jetson-tk1-2gb kernel: tegra-xusb 70090000.usb: Firmware timestamp: 2014-09-16 02:10:07 UTC
Jan 30 15:31:06 b3-jetson-tk1-2gb kernel: tegra-xusb 70090000.usb: xHCI Host Controller
Jan 30 15:31:06 b3-jetson-tk1-2gb kernel: tegra-xusb 70090000.usb: new USB bus registered, assigned bus number 3
Jan 30 15:31:06 b3-jetson-tk1-2gb kernel: tegra-xusb 70090000.usb: hcc params 0x0184f525 hci version 0x100 quirks 0x00010010
Jan 30 15:31:06 b3-jetson-tk1-2gb kernel: tegra-xusb 70090000.usb: irq 347, io mem 0x70090000
```

ch0mik, 2017-01-31:
Thx Robert, it works ;) !
PS. its similar problem on Debian`s Wiki : https://wiki.debian.org/InstallingDebianOn/NVIDIA/Jetson-TX1

closed 2017-04-28 by ch0mik