RobertCNelson/netinstall #60: stucking disk sata

ch0mik, 2017-02-17:
Hello,

I noticed that in my Jetson k1, the kernel newer than 4.9.6 (v4.9.6-ARMv7-lpae-x4) makes tapping disk drive sata.

PS. is v4.10.0-rc8-tegra-r0 for TK1 too ?

Regards
P.

RobertCNelson, 2017-02-17:
@ch0mik tapping? sounds more like a harddrive failure to me, run...

```
sudo smartctl -a /dev/sda
```

the tegra linux-image file is just the mainline tegra_defconfig build..

Regards,

ch0mik, 2017-02-17:
smartctl : http://pastebin.com/p6t4cyN6

RobertCNelson, 2017-02-17:
```
SMART overall-health self-assessment test result: FAILED!
Drive failure expected in less than 24 hours. SAVE ALL DATA.
No failed Attributes found.
```

Harddrives don't last forever, time to get a new one..

ch0mik, 2017-02-17:
Robert, Do You use a SSD HD for TK1 ?

PS. my new dmesg http://pastebin.com/0gurZ7qq

RobertCNelson, 2017-02-17:
I only have 4 spinning drives left in my build farm, been slowly replacing all spinning laptop drives with

https://www.amazon.com/gp/product/B009SX6VLC/

as they fail..

```
#wand quad
a1: ata1.00: ATA-8: ST9120310AS, SD03, max UDMA/133
a2: ata1.00: ATA-9: ADATA SP600, 5.8, max UDMA/133
a3: ata1.00: ATA-8: ST9120310AS, SD03, max UDMA/133
a4: ata1.00: ATA-8: ST9120310AS, SD03, max UDMA/133
a5: ata1.00: ATA-9: ADATA SP600, 5.8, max UDMA/133
a6: ata1.00: ATA-9: ADATA SP600, 5.8, max UDMA/133

#omap5_uevm
b1: ata1.00: ATA-9: ADATA SP600, 5.2, max UDMA/133
b2: ata1.00: ATA-9: ADATA SP600, 5.2, max UDMA/133

#Jetson TK1:
b3: ata1.00: ATA-9: ADATA SP600, 5.8, max UDMA/133
b4: ata1.00: ATA-9: ADATA SP600, 5.8, max UDMA/133

apm: ata1.00: ATA-8: Hitachi HCC543216A7A380, ES1OA60W, max UDMA/133
```

Regards,

ch0mik, 2017-04-22:
on kernel 4.9.23 works fine - without strange noise from disk ;)

ch0mik, 2017-04-28:
on 4.9.25 works fine too

closed 2017-04-28 by ch0mik