RobertCNelson/netinstall #63: Beaglebone Green Wifi connection

kriedel, 2017-04-28:
Hi,

I cannot connect to wifi network, with connman a connection is not possible:
connmanctl technologies
/net/connman/technology/wifi
  Name = WiFi
  Type = wifi
  Powered = True
  Connected = False
  Tethering = False

connmanctl scan wifi
Error /net/connman/technology/wifi: No carrier

connmanctl services
*AO Wired                ethernet_10bf482ff3d3_cable

dmesg delivers:
[   62.902610] usb 1-1: Product: 802.11n WLAN Adapter
[   62.902640] usb 1-1: Manufacturer: Realtek
[   62.902669] usb 1-1: SerialNumber: 00e04c000001
[   64.271023] rtl8192cu: Chip version 0x10
[   64.357525] rtl8192cu: MAC address: 64:70:02:29:fa:62
[   64.357564] rtl8192cu: Board Type 0
[   64.357671] rtl_usb: rx_max_size 15360, rx_urb_num 8, in_ep 1
[   64.357864] rtl8192cu: Loading firmware rtlwifi/rtl8192cufw_TMSC.bin
[   64.459963] ieee80211 phy0: Selected rate control algorithm 'rtl_rc'
[   64.470814] usbcore: registered new interface driver rtl8192cu
[   64.991984] rtl8192cu: MAC auto ON okay!

If I remove connman and use manual entries in /etc/network/interfaces a connection is possible but only to eth0 **or** wlan0. With both interfaces (auto eth0 and auto wlan0) I get a long boot time If I connect only one interface.

What's the right way to set up wifi, is it a wrong wifi adapter?

Regards,
Kai

RobertCNelson, 2017-04-28:
@kriedel that's why we use connman vs /etc/network/interfaces...

Regards,

kriedel, 2017-04-28:
OK, but what is the problem with connman and wifi?

Regards,
Kai

kriedel, 2017-04-28:
ubuntu 16.04.2 (ubuntu-16.04.2-console-armhf-2017-04-07.tar.xz)

RobertCNelson, 2017-04-28:
Yeah i see it too, i'm trying a build xenial build with connman disabled.

I don't test ubuntu very often, so i should leave xenial with the network defaults..

kriedel, 2017-05-11:
I try to use the new build based on the master branch, but the install script RootStock-NG.sh halt after the following output:

```
Log: (chroot) Warning, qemu can fail here... (run on real armv7l hardware for production images)
Log: (chroot): [git clone https://github.com/beagleboard/bb.org-overlays /opt/source/bb.org-overlays --depth 1 || true]
Cloning into '/opt/source/bb.org-overlays'...
remote: Counting objects: 161, done.
remote: Compressing objects: 100% (121/121), done. 
```
No image is generated in the deploy folder.

RobertCNelson, 2017-05-11:
@kriedel that's a known issue, hence the warning:

```
Log: (chroot) Warning, qemu can fail here... (run on real armv7l hardware for production images)
```

Regards,

kriedel, 2017-05-21:
It's OK now with the new build, but the source list seems now to be lost:

`sudo apt-get update
0% [Connecting to ports.ubuntu.com (172.1.8.1)] [Connecting to repos.rcn-ee.com
...
Err:3 http://repos.rcn-ee.com/ubuntu xenial InRelease                           
  Could not connect to repos.rcn-ee.com:80 (172.1.8.1), connection timed out   `

In which file I have to correct the source list?

Regards,
Kai

RobertCNelson, 2017-05-22:
ah that looks like network problems..

kriedel, 2017-05-22:
ping of http://repos.rcn-ee.com/ubuntu is OK...