RobertCNelson/netinstall #64: Fresh install of Debian Stretch uses kernel 4.4, instead of 4.9

kscz, 2017-06-26:
A fresh install of Debian Stretch via netinstall on an x86_64 machine yields the following:
```
kscz@debian:~$ uname -a
Linux debian 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u1 (2017-06-18) x86_64 GNU/Linux
```
Whereas a fresh install of Debian Stretch on my Beaglebone Green Wireless yields:
```
kscz@debian:~$ uname -a
Linux debian 4.4.68-ti-r110 #1 SMP Wed Jun 21 05:28:34 UTC 2017 armv7l GNU/Linux
```
Considering the fact that all the way up to `linux-image-4.9.33-ti-r42` is there some reason why a closer version of the Linux kernel isn't available?

Also: if this is the wrong venue for this issue, please let me know and I'll re-cut it somewhere more appropriate!

Thank you for any help!

RobertCNelson, 2017-06-26:
@kscz yeah it would make sense to allow users to easily override 4.4.x.

https://github.com/RobertCNelson/netinstall/commit/00474eb6412094ad082b3f752e347d38f2f67a81

Specially since the next lts (4.14.x is probably 2 months away from 4.14.0-rc1)..

The main reason for the 4.4.x default is mostly because of the blue, when we port librobotics to v4.9.x-ti i'll make the jump for the bone family.

Regards,

kscz, 2017-06-26:
Wow, thanks for the quick response!  ...and fix!  I assume this also pulls over the latest version of the linux-image-firmware-4.9.x as well?

kscz, 2017-06-26:
I realized I shouldn't close this, because it's not the default yet!

As a side question, is there something I should be doing to get the [BB-CAPE-DISP-CT43](https://www.element14.com/community/docs/DOC-81966) working? I was under the impression based on the discussions in google group that 4.9 would have all the necessary magic to make that screen work... is that not the case?  I'm still seeing:
```
bone_capemgr bone_capemgr: loader: failed to load slot-3 BB-CAPE-DISP-CT4:00A0 (prio 0)
```
The screen appears to be working when I use the BBGW default image, but doesn't work when I use the netinstall method.

I know this is *definitely* not the right place for this question, so let me know if I should hop on IRC or something!

RobertCNelson, 2017-06-26:
@kscz to get that cape working two things:

Install "bb-cape-overlays" https://github.com/beagleboard/bb.org-overlays/

It's packaged:

```
sudo apt update ; sudo apt install bb-cape-overlays
```
Then in /boot/uEnv.txt
```
enable_uboot_overlays=1
```

Double check by connecting a up a usb-serial adapter to the on board serial debug header and setup xorg.conf like so:

https://eewiki.net/display/linuxonarm/BeagleBone+Black#BeagleBoneBlack-HDMI

Regards,

closed 2024-09-26 by kscz