RobertCNelson/netinstall #65: Jetson "zesty" card boots part way and does nothing.

sarnold, 2017-07-20:
Hey this is the first time I tried this one so I'm not exactly sure what it's *supposed* to do, but all it has is a kernel and initramfs which I assume is the installer?  I get the "using serial interface" message, but then nothing.

What exactly am I waiting for?  Should I get a console/installer interface at some point?  Isn't there a newer (mainline) u-boot that works now?

Thanks...

```U-Boot SPL 2014.10-rc2-00001-g9f88c9e (Dec 01 2014 - 14:29:15)

U-Boot 2014.10-rc2-00001-g9f88c9e (Dec 01 2014 - 14:29:15)

TEGRA124
Board: NVIDIA Jetson TK1
I2C:   ready
DRAM:  2 GiB
MMC:   Tegra SD/MMC: 0, Tegra SD/MMC: 1
*** Warning - bad CRC, using default environment

tegra-pcie: PCI regions:
tegra-pcie:   I/O: 0x12000000-0x12010000
tegra-pcie:   non-prefetchable memory: 0x13000000-0x20000000
tegra-pcie:   prefetchable memory: 0x20000000-0x40000000
tegra-pcie: 2x1, 1x1 configuration
W֖▒▒pcie: probing port 0, using 2 lanes
tegra-pcie: link 0 down, retrying
tegra-pcie: link 0 down, retrying
tegra-pcie: link 0 down, retrying
tegra-pcie: link 0 down, ignoring
tegra-pcie: probing port 1, using 1 lanes
In:    serial
Out:   serial
Err:   serial
Net:   RTL8169#0
Warning: RTL8169#0 using MAC address from net device

Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1...
Found U-Boot script /boot/boot.scr
4782 bytes read in 74 ms (62.5 KiB/s)
## Executing script at 90000000

Searching for: /boot/uEnv.txt ...
Trying: mmc 1:1 ...
Found: /boot/uEnv.txt on [mmc 1:1] ...
206 bytes read in 88 ms (2 KiB/s)
Using: [dtb=tegra124-jetson-tk1.dtb] ...
Using: [uname_r=current] ...
Warning: [uuid] is not set in /boot/uEnv.txt ...
Warning: [mmcroot] is not set in /boot/uEnv.txt...
Using: [root=/dev/mmcblk0p1 ro] ...
Warning: [rootfstype] is not set in /boot/uEnv.txt...
Using: [rootfstype=ext4] ...
Warning: [console] is not defined...

Searching for: Linux Kernel [vmlinuz-current] ...
Trying: /boot/vmlinuz-current ...
Found: /boot/vmlinuz-current ...
6933856 bytes read in 534 ms (12.4 MiB/s)

Searching for: Device Tree Blob [tegra124-jetson-tk1.dtb] ...
Trying: /boot/dtbs/current/tegra124-jetson-tk1.dtb ...
Found: /boot/dtbs/current/tegra124-jetson-tk1.dtb ...
70095 bytes read in 214 ms (319.3 KiB/s)

Searching for: Linux initial RAM disk [initrd.img-current] ...
Trying: /boot/initrd.img-current ...
Found: /boot/initrd.img-current ...
73489522 bytes read in 5005 ms (14 MiB/s)

[... 341 lines cut ...]
[   10.561007] hub 3-0:1.0: 6 ports detected
[   10.566085] tegra-xusb 70090000.usb: xHCI Host Controller
[   10.571537] tegra-xusb 70090000.usb: new USB bus registered, assigned bus number 4
[   10.579291] usb usb4: We don't know the algorithms for LPM for this host, disabling LPM.
[   10.587539] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003
[   10.594361] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[   10.601613] usb usb4: Product: xHCI Host Controller
[   10.606493] usb usb4: Manufacturer: Linux 4.9.38-armv7-lpae-x4 xhci-hcd
[   10.613136] usb usb4: SerialNumber: 70090000.usb
[   10.618892] hub 4-0:1.0: USB hub found
[   10.622711] hub 4-0:1.0: 2 ports detected
[   10.628581] sdhci-tegra 700b0400.sdhci: Got CD GPIO
[   10.633533] sdhci-tegra 700b0400.sdhci: Got WP GPIO
[   10.638610] mmc1: Unknown controller version (3). You may experience problems.
[   10.646029] mmc1: Invalid maximum block size, assuming 512 bytes
[   10.662248] ata1: SATA link down (SStatus 0 SControl 300)
[   10.719354] mmc1: SDHCI controller on 700b0400.sdhci [700b0400.sdhci] using ADMA
[   10.727146] tegra-pcie 1003000.pcie-controller: 2x1, 1x1 configuration
[   10.735160] tegra-pcie 1003000.pcie-controller: probing port 0, using 2 lanes
[   10.744658] tegra-pcie 1003000.pcie-controller: Slot present pin change, signature: 00000008
[   10.838289] mmc1: new high speed SDHC card at address e624
[   10.844714] mmcblk1: mmc1:e624 SD32G 29.7 GiB
[   10.854293]  mmcblk1: p1
[   10.918835] usb 3-3: new full-speed USB device number 2 using tegra-xusb
```

RobertCNelson, 2017-07-20:
Hi @sarnold ,

yeah you should get a ncures based terminal window over the serial interface.

from my notes, https://github.com/RobertCNelson/netinstall/blob/master/hwpack/tegra124-jetson-tk1.conf#L28

i last used v2017.03-rc1...

Sadly both my jetson's are now in a box un-plugged.  Ran into random stability issues when running under 24/7 100% load.  One would run for weeks, the other a couple hours..

on the bright side, it's easier for me to boot test. ;)

Regards

RobertCNelson, 2017-07-20:
btw, double check with "--distro stretch" that was the last distro i personally used on the jetson... the zesty enablement was merged by another user, haven't had to chance to personally test it yet.  I just assumed it worked for them.  ubuntu might have thrown in a new curve ball in their debian-installer script, that i'd have to work around..

Regards,

sarnold, 2017-07-20:
At the end of the sdcard script it says:

-----------------------------
Required: https://github.com/NVIDIA/tegra-uboot-flasher-scripts (tested: u-boot v2017.03-rc1)
-----------------------------

Required to be where and for what?  Is there a manual step I'm missing?  I'll test it with stretch now...

RobertCNelson, 2017-07-20:
That's just the version of u-boot mainline i had used.

at one point it, it was the first to use config_distro_default, hence the required..

Regards,

sarnold, 2017-07-21:
Stretch stops at the same place:

```
[    9.670528] usb 3-3: new full-speed USB device number 2 using tegra-xusb
```

Got any suggestions?  It does boot with my Gentoo sdcard, but at some point the Gentoo root I put on the flash got hosed up :/

sarnold, 2017-07-21:
So I went back a made a card like I did for Gentoo using your ubuntu rootfs; turns out I used the tegra-uboot-scripts to make it boot with your tegra kernel.  Since it looks like it still has the original u-boot in flash:

U-Boot SPL 2014.10-rc2-00001-g9f88c9e (Dec 01 2014 - 14:29:15)

I should probably flash a newer u-boot if I want this to work correctly?