RobertCNelson/netinstall #66: uEnv.txt

Antonstr, 2017-08-30:
If you wrongly edited the uEnv.txt on a BeagleBoneBlack and the Beagle boots into initramfs, is there any way you can recover the beagle 

##Disable HDMI
cape_disable=capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN

##Disable eMMC
cape_disable=capemgr.disable_partno=BB-BONE-EMMC-2G

The problem appears to be the removal of the # at Disable eMMC
I can access the uEnv.txt file but it does not allow me to edit it.

Any advice will be greatly appreciated
Thank you

RobertCNelson, 2017-08-30:
@Antonstr sure just boot a console image on a microSD, mount the eMMC and fix it:

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Jessie_Snapshot_console

Regards,

Antonstr, 2017-08-31:
> Snapshot_console

Good day Robert

Thank you after 3 reboots I could eventually fix it

Regards

Anton

shameless, 2018-01-15:
Hey,  after replace the kernel of the Debian, my Beagle boots into initramfs too, and I don't know how to fix it ,maybe I should edit uEvn.txt to make it right.