RobertCNelson/netinstall #70: Installer installs uboot on the wrong storage on beaglebone.

staviq, 2019-06-20:
I made a netinst sdcard, used the sdcard to install system on beaglebone and the system is not bootable, bootloader is missing because the installer installed in on the sd card instead of the mmc.

Also, the installer wanted to overwrite itself on the sdcard, i had to manually set the partitioning so it uses mmc.

staviq, 2019-06-20:
If i remove the netinstall sdcard right after it boots, since netinst runs from ram, the automatic partitioning looks good, and the install finishes, but the system is still missing the bootloader.

RobertCNelson, 2019-06-20:
@staviq this is expected, what ever media/device is used with mk_mmc.sh has to be installed on the device thru out the life of the install.

I wrote this script to install two may install types:

Single Device
1: mmc install

Dual Device
1: mmc (bootloader/kernel) install
2: USB/SATA/etc Hard Drive (rootfs) Install

For what you want, I'd have to rewrite the logic behind this script.

Regards,

staviq, 2019-06-21:
Ok, i get it now.

And it worked when i booted beagle from sd and created the installer on the mmc, and then when i booted from the netinstall on the mmc, the installer did install correctly.

It is however very confusing that the script pretty much creates an one time install that overwrites itself, at least on the beaglebone.

I think You should either put this information in the description somewhere, literally saying that the netinstall is suppsed to be put on the same target you want the system on.

Or, possibly, to add a prompt in the installer that lets you select target device.