RobertCNelson/netinstall #73: questions, repo status

sarnold, 2021-04-16:
Howdy:

Seems like a dearth of info/tools for Jetson tk1 flashing (mainly u-boot/emmc) and the last(?) official L4T bsp tools really did a number on the emmc (stopped it from booting *and* broke ethernet when I boot from sdcard).  I didn't even know you had this until today, so I made a card and I'm not sure what the note at the end means:

```
-----------------------------
Finished populating Boot Partition
-----------------------------
mk_mmc.sh script complete
Script Version git: 4ef52f2
-----------------------------
Required: https://github.com/NVIDIA/tegra-uboot-flasher-scripts (tested: u-boot v2017.03-rc1)
-----------------------------
```

Does that mean I need to manually use the (really stale) tegra-uboot-flasher-scripts *after* running ``mk_mmc.sh``?  I thought I saw it download u-boot/kernel stuff in my console; am I supposed to flash whatever your script downloaded or do I go back and build from u-boot source?

I did see your answer to issue #72 but I'd really like to get this thing back to a bootable state; I'm willing to make a PR with some updates but that requires picking your brains a bit...  Thanks!

RobertCNelson, 2021-04-17:
Hi @sarnold a newer version 'should' work, that was just the last version i had tested..  At the time of the commit, i had 2 working tegra's, they probally ran for a number of years after that commit.

Looking at my hwpack file:

https://github.com/RobertCNelson/netinstall/blob/master/hwpack/tegra124-jetson-tk1.conf

```
#Bootloader: u-boot features:
conf_config_distro_defaults="enable"
conf_smart_uboot="enable"
```

on the tegra, it loads this file:

https://github.com/RobertCNelson/netinstall/blob/master/lib/distro_defaults.cmd

https://github.com/RobertCNelson/netinstall/blob/master/mk_mmc.sh#L1120-L1122

as /boot/boot.scr

i think tegra supportrs /boot/extlinux/extlinux.conf now, that would be the better path going forward. ;)

Regards,

RobertCNelson, 2021-04-17:
Ah, so i must have 'flashed" u-boot v2017.03-rc1 with the tegra-uboot-flasher-scripts

sarnold, 2021-04-17:
Yup, the tegra flasher thing builds u-boot and it apparently defaults to "latest" so I flashed it and now it boots my gentoo card again:

```
$ uname -a
Linux genson 4.19.63-tegra-r0 #2 SMP PREEMPT Sat Aug 3 18:32:25 PDT 2019 armv7l ARMv7 Processor rev 3 (v7l) NVIDIA Tegra SoC (Flattened Device Tree) GNU/Linux
```

The L4T stuff just kernel panics, and tegra-5.11 (current) boots your installer but ethernet is broken and I'm not sure why yet.  (note I had to change your flasher ``kernel_subarch`` to tegra and used the stable kernel arg)

RobertCNelson, 2021-04-18:
Grab an old network switch, force it to operate at 100MB/s, last i remember there was an old bug on at 1GB/s network connection..