RobertCNelson/netinstall #74: Beagleboardxm - not able to boot distro buster

mahajanabhishek, 2022-11-17:
Used following command: 
_sudo ./mk_mmc.sh --mmc /dev/sdc --dtb omap3-beagle-xm --distro buster_

After successful writing of sd card. Not able to boot on beagleboard-xm. 
Getting following:
_spl_load_image_fat_os: error reading image args, err - -2_

Did I missed something ?

[netinstall.logs.txt](https://github.com/RobertCNelson/netinstall/files/10034131/netinstall.logs.txt)

mahajanabhishek, 2022-11-17:
Tried with --distro stretch --use-stable-kernel

Same issue on booting:
CU-Boot SPL 2018.03-00001-g21a26066ea (May 21 2018 - 15:23:05 -0500)
Trying to boot from MMC1
.ͱ}����}�����}���}ѽ�: error reading image args, err - -2

mahajanabhishek, 2022-11-18:
Going through old mail chain it seems this issue is could be related to microSD card. Although I tried different sdcard which worked with old version not working on new images. From old post of July 2020 I can see there was patch available (https://github.com/balena-os/balena-beaglebone/blob/29a9134e1ef339a1d095bac29845926ab24d4082/layers/meta-balena-beaglebone/recipes-kernel/linux/linux-beagleboard-5.4/0001-card-power-cycle.patch). which worked for some people but I will try and update.

RobertCNelson, 2022-11-18:
Sorry, the real issue.. I don't use any xM's 24/7 like i did previously, so when things break... yeah.. bit rot..